vhda/verilog_systemverilog.vim #178: Can set indent like below?

kdurant, 2019-03-04:
Now default indent is 
```verilog
module name
    (
    );
```
I want this:
```verilog
module name
(
);
```

vhda, 2019-03-04:
let g:verilog_disable_indent_lst="module,eos"

This does it, but it affects more than just the module declaration. Please test it and tell me if it is acceptable.

kdurant, 2019-03-05:
It no effect.
I fork and change
```viml
    if l:line =~ s:vlog_module
      return s:GetContextStartIndent("module"    , l:lnum) + l:open_offset
```  
to
```viml
    if l:line =~ s:vlog_module
      return 0
```

vhda, 2019-03-06:
I was using `=` to indent the example you provided, but today I noticed that the behavior was not what I expected in Insert mode. This is fixed in master.

All test cases are passing, so let's hope the code wasn't there for a reason not included in the tests! :)

commit 3cc5147a (2019-03-06): Indent: Remove module from block declaration pattern

This would only affect indentation when in Insert mode and would force
indentation after a module declaration even if g:verilog_disable_indent_lst
included "module".
Fixes #178
closed 2019-03-06 by vhda