vhda/verilog_systemverilog.vim #180: how to enable Omni-completion

cqhdxx, 2019-03-12:
When I generate the tags with ctags from the help doc:
1. Module instantiation port names.
2. Function/task arguments.
3. Object methods and attributes.

In order to use omni completion a tags file must be generated using the
following arguments:

* `--extra=+q` - Enable hierarchy qualified tags extraction.
* `--fields=+i` - Enable class inheritance extraction.
* `-n` - (Optional) Use line number instead of Ex: patterns to identify
declaration (generates smaller tags file).

It's can not work if a `.` character is input to the terminal. But when I use Ctrl+P or Ctrl+N，it's can auto complete the words.

vhda, 2019-03-14:
Are you using `CTRL-X_CTRL-O` to invoke omni-completion?

cqhdxx, 2019-03-18:
I use a  . character to invoke omni-completion, it's doesn't work.

vhda, 2019-03-20:
Go inside this plugin, into "test" folder.
Open "test.v".
Go to end of line 12 and try invoking omni completion.

cqhdxx, 2019-03-21:
Follow your instructions and press CTRL-X_CTRL-O, and I can achieve automatic completion. And press a . character it's doesn't work. 
Is the omni-completion only work for Module instantiation?

vhda, 2019-03-23:
The limitations of the current implementation of omni completion are described in both README.md and Vim's internal help: `:help verilog-omni`.
I also suggest that you read the part about supertab included in the README.md file.

vhda, 2019-06-07:
Any updates?
Did the documentation help?

leixianb, 2019-06-26:
I have the same problem. And I read the doc, tried to fix it, but failed. sentence in Readme.md: " This will only work if a . character is found in the keyword behind the cursor" is not easy to understand. could you please give an example?

vhda, 2019-07-02:
Sorry for the late reply.

Could I ask you to make the same test as I asked @cqhdxx ?

> Go inside this plugin, into "test" folder.
> Open "test.v".
> Go to end of line 12 and try invoking omni completion.

That is, pressing CTRL-X_CTRL-O in insert mode after the '.' at the end of line 12.

cqhdxx, 2019-07-04:
@vhda Sorry for the late reply, It's really helpful for me to use this vimscript.

vhda, 2019-07-26:
@cqhdxx did my instructions work?

vhda, 2020-03-07:
Marking as resolved. Please feel free to reopen if necessary.

closed 2020-03-07 by vhda