vhda/verilog_systemverilog.vim #183: Line comment with a comma highlight as code

toastedcornflakes, 2019-05-31:
Hello! 

The following snippet gets the wrong colour after the comma, as if the line-comment was ended by the comma.

``` assign mysig = condition ?
                // Comment, with comma
                a : b;```

![comment_vim](https://user-images.githubusercontent.com/6625812/58698603-7356a480-8393-11e9-936e-47e2ccaec7f8.PNG)

How can I debug this further?
Many thanks

closed 2019-06-06 by vhda

vhda, 2019-06-06:
I had a commit laying around that apparently fixes this.
Would you be so kind a confirm that this is fixed?

toastedcornflakes, 2019-07-21:
Brilliant, it's indeed fixed. Many thanks, and sorry for the late reply, I somehow missed the notification. Thanks again!

commit ae5bf6d1 (2019-06-06): Syntax: Skip comments in all region syntax entries

Fixes #183