vhda/verilog_systemverilog.vim #187: the keyword endmodule is not highlight

puxiancheng, 2019-08-16:
when i open a .v file though vim, i type G go to the end of the file, the keyword endmodule is not highlight.
if i type ctrl+f go to the end of the file, the keyword endmodule is highlight.

vhda, 2019-08-25:
Hi @puxiancheng,

Please try adding the following to your vimrc:
```vim
autocmd FileType verilog_systemverilog syntax sync fromstart
```
This will make vim always defines syntax by parsing the complete file. Unfortunately, I expect that this will make the syntax parsing slower because it has to parse more items. But test to see if it becomes too slow or not.

puxiancheng, 2019-09-06:
thanks!!!
it works, and it does not become too slow.

closed 2019-09-06 by puxiancheng