vhda/verilog_systemverilog.vim #190: Intendation at end of medium-large files is slow

engr248, 2020-01-22:
On a decent sized file of SystemVerilog, the indentation performance slows down considerably towards the end of the file. I noticed this on an SV file that's ~1800 lines long and primarily consists of instantiations of other modules (it's the top level of an SoC instantiating a processor and peripherals).

When working near the top of the file, inside the module, the autoindent performance is acceptably fast. When working near the bottom of the file, still inside the module, the autoindent slows down to a crawl to where beginning a new comment block immediately before the endmodule takes approximately 3 seconds to figure out where to indent to. While it's figuring this out I can still type, but nothing shows up until it figures out the indent and catches up.

Setting g:verilog_verbose = 1 and looking at messages as they scroll by and in :messages shows me that GetContextIndent is going backwards line by line through the entire file until it hits the module instantiation, figures out it's in a module, and then indents - this is what is taking seconds.

Looking through the configuration documentation I didn't see any easy way to tweak the performance of indentation. Even disabling 'module' for indentation didn't change it as, I believe, the script still has to determine it's in a module, and that's what's taking several seconds.

Any advice?

vhda, 2020-01-28:
Give me some time to look into this (currently swamped in work and personal life :)

vhda, 2020-03-07:
Could you please try the branch [indent/stop_at_instance](https://github.com/vhda/verilog_systemverilog.vim/tree/indent/stop_at_instance)?

I'm storing the indent level of `);` and using it if I then find a `.something` preceding it.

engr248, 2020-03-09:
It's definitely faster, but the the current state of the indent/stop_at_instance branch does not produce the same results as master for me.

Example:
```systemverilog
module blah #( 
    //parameters omitted
) (
    //ports omitted
);
    //~2000 lines omitted
    ModuleName #( 
        //parameters omitted
    ) InstanceName (
        .portName(connection)
    );
    //note that the above ); is at indent level of 4 spaces
endmodule:blah
```

On master, if I am on the end of the last `);` in insert mode and hit enter it will churn for a few seconds and then put the cursor on the next line at indent level of 4 spaces (as previously reported).

On indent/stop_at_instance, if am on the end of the last `);` in insert mode and hit enter it will immediately go to the next line at indent level 0, which is not what I expected. Output of :messages with verbose enabled below.
```
GetContextIndent:2139:     );
Found possible end of instance on line 2139
GetContextIndent:2138:         .portName(connection)
Found instance at line 2138. Returning previously stored indent level of 0
```
Note that for this test I simply opened the file, hit `G` to go to the bottom, moved to the `);` in normal mode, and then went to insert mode in case that has an effect on the operation.

engr248, 2020-03-09:
I appreciate the work and am glad to test out again as needed!

vhda, 2020-03-09:
Could you please try again?

engr248, 2020-03-31:
Ahha! That branch now shows correct and quick behavior for me. With verbose on the messages show it returning the correct indent level of 4 for my test case as well.
Cheers, and thanks!

vhda, 2020-04-01:
Merged. Cheers!

closed 2020-04-01 by vhda

netvolcano, 2023-02-28:
I found a similar issue with the plugin of 3.0 version.  When I have more 2 thousands of interface signal, there will be a long time waiting to finish GetContextIndent function ( I used a profile log file to confirm this). How can I fix this issue? I tried the current version, but it can not slove the problem. Can you give me any clues?

netvolcano, 2023-02-28:
Can we a limit of the max scan line number?

PR #192 (2020-04-01): Indent: Stop searching for indentation at end of instance
Fixes #190