vhda/verilog_systemverilog.vim #191: Unhighlighted  pin after linebreak (\n)

Strongsaxophone, 2020-03-13:
Consider having a module : 
 alu ALU(.in(), .in1(), .operation(), 
               .out());
the .out() is not highlighted.
You can also view [this](https://ibb.co/j3HWPdF) picture to get what I mean.

vhda, 2020-03-22:
Hi @Strongsaxophone,

You stepped on a syntax bug, which I never noticed before because I always separate the port name from the parentheses. The bad news is that my intention was to **not** highlight port names.
The ports are currently being highlighted using the verilogMethod syntax item that was created to highlight function and task calls in the form `object.function()`. As a side benefit, I've kept its capability of highlighting any keyword immediately followed by a parentheses, but excluding specifically ports in instances. The idea was to have the module name of the instance highlighted, but keeping everything else "clean".
If this is something you really want to keep I can work on creating a syntax item for that specific purpose. Or you can include it in `g:verilog_syntax_custom` as a personal setting.

Thanks!

Strongsaxophone, 2020-03-22:
Hello @vhda,
I will change that for myself with g:verilog_syntax_custom . I think that's a good idea to have pins in another color different from module instances.
Best Regards.

vhda, 2020-03-24:
I'm open to improvement suggestions, so let's keep this open. Free time is not much lately, but I'll take this into account while fixing the bug you uncovered.
Are you ok with the current highlight, or would you prefer another "color type"?

Strongsaxophone, 2020-03-24:
To be honest I have no idea how hard it ( I do not know vim scripting) is to change colors of different parts of Verilog code, But I as I see there are three kind of pins which is a good idea to have different colors : 
1. parameters ( which is not hard to identify as it is in #() block, )
2. instance names
3. instance pins
So it is a good idea that different pins ( 3 types pointed above),  has different colors, but I do not know how hard it is to make these changes :) .