vhda/verilog_systemverilog.vim #193: issue with syntax hilight with inline/multi-line comments

justrajdeep, 2020-04-20:
Hi 

This plugin is amazing. I cannot thank you enough.

I see only a trivial issue with syntax highlight.

    /* uvm_analysis_port #(REQ) mon_analysis_port;  //TLM analysis port */

if i have something like the above line; it misses the multi line comment closure(`*/`) because of the inline comment(`//`).

Can you please take look?

kirsa, 2020-04-27:
Also seeing this issue. Would really appreciate any pointers on how to resolve this.

kirsa, 2020-04-27:
My local fix for this is in ~/.vim/bundle/verilog_systemverilog.vim/syntax/verilog_systemverilog.vim add the following after line 357 :
syn region verilogComment start="/\*" end="/*\"

vhda, 2020-04-27:
Could you please try again?

closed 2020-04-27 by vhda

kirsa, 2020-04-28:
Works for me, thanks!

commit c63b5ee8 (2020-04-27): Syntax: Don't skip comments inside comments

Fixes #193