vhda/verilog_systemverilog.vim #199: Demo Request

BharadwajSP, 2020-06-11:
Hi, 
   This is absolute must-have plugin for Verification engineers. I could not find any demo of all the features available in it. it will be a great help for newcomers to take advantage. please point to me if anything available elsewhere. 

thanks 
Keshava

vhda, 2020-06-12:
Hi Keshava,

What kind of demo would you be looking for?
Most of the information available on this plugin is in its [README.md](https://github.com/vhda/verilog_systemverilog.vim/blob/master/README.md) and the included Vim help text (`:help verilog_systemverilog.txt`). Is this not sufficient?

Best regards,
Vitor

BharadwajSP, 2023-04-21:
Hi Vitor, 

   I'm sorry I didn't check back on the comment/request I made earlier. I am looking for a Video Demo of most features specific to the SystemVerilog. for example, I have no idea how OMNI completion, error format follow instance will work. maybe because I never got it to work.! 
      plus, when the project gets bigger at the cluster and SOC level where you have thousands of files or an individual file with thousands of lines of code( RAL files) it takes a lot of time to open itself with syntax fold on. 
     So, is it possible to show us the features with DEMO and how to set it up, May be taking an example of UVM LIB and explaining all these would be a great help? 

Thanks 
keshava.