vhda/verilog_systemverilog.vim #203: default indent (question)

brayden-jo, 2021-02-25:
Thank you for making a good plugin. 
I have a simple question, I review the document but I didn't find a proper solution yet. 

This plug-in seems to use default 4-space of indent. 
Can I turn-off this feature ? 
We usually the coding style which starts with column 0. Please check below two coding style.  

```
// current
module (
    clk, 
    reset_n,
    i_data,
    o_data
);

    input clk;
    input reset_n;

endmodule
```

```
// new
module (
    clk, 
    reset_n,
    i_data,
    o_data
);

input clk;                               // no indent
input reset_n;

always @ (posedge clk or negedge reset_n) begin
    if (!reset_n) begin
        o_data <= 4'b0;
   end else begin
       o_data <= i_data;
   end
end

endmodule
```

brayden-jo, 2021-02-25:
I added below line to .vimrc but it does not works. 

```
let g:verilog_disable_indent_lst = "module,class,interface"
```

vhda, 2021-02-27:
Hi @brayden-jo,

That should work exactly as you need if you also include "eos" in the comma separated list.
And the plugin will follow by default your `'shiftwidth'` setting.
This what I get with `set sw=2` and `let g:verilog_disable_indent_lst="eos"`:
```systemverilog
// new
module (
  clk,
  reset_n,
  i_data,
  o_data
);

  input clk;                               // no indent
  input reset_n;

  always @ (posedge clk or negedge reset_n) begin
    if (!reset_n) begin
      o_data <= 4'b0;
    end else begin
      o_data <= i_data;
    end
  end

endmodule
```

This what I get with `let g:verilog_disable_indent_lst="eos,module"`:
```systemverilog
// new
module (
  clk,
  reset_n,
  i_data,
  o_data
);

input clk;                               // no indent
input reset_n;

always @ (posedge clk or negedge reset_n) begin
  if (!reset_n) begin
    o_data <= 4'b0;
  end else begin
    o_data <= i_data;
  end
end

endmodule
```
Please try disabling other plugins and/or settings that could be affecting this behavior.

Hope this helps,
Vitor

brayden-jo, 2021-03-02:
@vhda ,
Thanks you for your reply. 

When I set the below setting within the vim, it works well.

let g:verilog_disable_indent_lst = "module,class,interface"

But as I mentioned before, it doesn't work with .vimrc file. 
I delete all plugins and use the simplest .vimrc as below:

```
call plug#begin('~/.vim/plugged')
    Plug 'vhda/verilog_systemverilog.vim'
call plug#end()

let g:verilog_disable_indent_lst="eos,module"
```

Currently I use the this command in the vim to apply the `let g:~~~~` configure.

source ~/.vimrc

vhda, 2021-03-02:
Instead of sourcing vimrc just open vim and run the following command:

```vim
:let g:verilog_disable_indent_lst
```
Does this show the value you defined in the vimrc?
Then try opening a verilog file and check this command again.
Could you also confirm that filetype value in the verilog file?
```vim
:set filetype
```

brayden-jo, 2021-03-04:
@vhda ,
when I run the vim and run the following command:

```
:let g:verilog_disable_indent_lst
```

default options is shown as below 

```
g:verilog disable indent lst eos
```

when i run the `:set filetype`, I can see the 'filetype=verilog systemverilog'

Thanks.

brayden-jo, 2021-03-04:
[image]

vhda, 2021-03-08:
Hi @brayden-jo,
Sorry for the delayed reply.

Are you saying that `g:verilog_disable_indent_lst` variable does not work in .vimrc. But what about setting it dynamically? That is, if you open a Verilog file and set this variable, do you see indentation start to work correctly?
I'm running out of ideas. Everything looks good and it should be working!

brayden-jo, 2021-03-08:
[image]

@vhda ,

dynamically setting doesn't work too for the *v/sv file.

vhda, 2021-03-21:
I've just tried on my side with a clean .vimrc and everything is working as expected. I can try it in CentOS during this week just to confirm, but I'm really confident that shouldn't be the issue.
Using the example file above, try the following:
```vim
gg=G
```
This will indent the complete file.
This is the result with default settings:
[image]
Then I set the variable to "eos,module" and get this: after re-indenting:
[image]

closed 2021-09-06 by vhda

netvolcano, 2021-11-16:
I had the same issue and found a way to solve it. I suggest to add the following lines in the `plugin\verilog_systemverilog.vim`. When I put my setting the `.vimrc` file, this following statements will not overwrite my setting.

 ```
" Global plugin settings
if !exists("g:verilog_disable_indent_lst")
    let g:verilog_disable_indent_lst="eos"
endif
 ```