vhda/verilog_systemverilog.vim #204: Rainbow 'begin/end' Support

thuyng88, 2021-02-28:
Thanks for the awesome plugin. I've tried to match begin/end with matching colors just like rainbow plugin does for parentheses.
[image]
However even I changed the parentheses setting in rainbow it still doesn't work 
[image]
[image]
I believe this is due to syntax priority between verilog_systemverilog and rainbow. What should I do to solve this issue? Thanks.

vhda, 2021-03-08:
I can't find a way of disabling this highlight on my side, as it is.
I've looked at some of those rainbow plugins and some have an "after" option that could be used to overcome this. Could you give it a try?

netvolcano, 2021-11-16:
Try to add the following lines into you rainbow setting (i.e. put them into `let s:rainbow_conf = {...}` )
```
\    'verilog_systemverilog': {
\        'parentheses': [
\              'start=/(/ end=/)/ fold',
\              'start=/\[/ end=/\]/ fold',
\              'start=/{/ end=/}/ fold',
\              'start=/\<begin\>/ end=/\<end\>/',
\         ],
\        'after': [
\              'syn clear verilogBeginEnd',
\              'syn match verilogOperator "[&|~><!*#%@+/=?:;}{,.\^\-\[\]]" ',
\              'syn region verilogComment start="/\*" end="\*/" contains=verilogTodo,@Spell ',
\              'syn match  verilogComment "//.*" contains=verilogTodo,@Spell ',
\           ],
\    },
```

vhda, 2023-05-10:
@tnguyen-tx did @netvolcano solution work for you?

netvolcano, 2023-05-11:
[image]

With my setting in the rainbow plugin.

real-bird-kun, 2023-12-22:
> Try to add the following lines into you rainbow setting (i.e. put them into `let s:rainbow_conf = {...}` )
>
> ```
> \    'verilog_systemverilog': {
> \        'parentheses': [
> \              'start=/(/ [...]

Hi~ I am trying to use your solution, but it doesn't work for me. The error info is in the following picture. Would you please help me figure it out?
[image]

My configuration is:
[image]

Thank your for your help in advance :)

netvolcano, 2023-12-23:
Have you installed the the verilog syntax plugin as below?
Plugin 'vhda/verilog_systemverilog.vim'

If it will not take effects after you installed the plugin, you need delete a line of "syn keyword verilogLabel       begin end fork join" from the verilog_systemverilog.vim file in the syntax directory.

IMPORTANT NOTICE
The contents of this email and any attachments may be privileged and confidential. If you are not the intended recipient, please delete the email immediately. It is strictly prohibited to disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.  ©Visinex Technology Co., Ltd copyright and reserve all rights. 
重要提示：本邮件（包括任何附件）可能含有专供明确的个人或目的使用的机密信息，并受法律保护。如果您并非该收件人，请立即删除此邮件。严禁通过任何渠道，以任何目的，向任何人披露、储存或复制邮件信息或者据此采取任何行动。感谢您的配合。 ©为旌科技有限公司 版权所有并保留一切权利。
 
From: real-bird-kun
Date: 2023-12-22 23:07
To: vhda/verilog_systemverilog.vim
CC: netvolcano; Mention
Subject: Re: [vhda/verilog_systemverilog.vim] Rainbow 'begin/end' Support (#204)
Try to add the following lines into you rainbow setting (i.e. put them into let s:rainbow_conf = {...} )
\    'verilog_systemverilog': {
\        'parentheses': [
\              'start=/(/ end=/)/ fold',
\              'start=/\[/ end=/\]/ fold',
\              'start=/{/ end=/}/ fold',
\              'start=/\<begin\>/ end=/\<end\>/',
\         ],
\        'after': [
\              'syn clear verilogBeginEnd',
\              'syn match verilogOperator "[&|~><!*#%@+/=?:;}{,.\^\-\[\]]" ',
\              'syn region verilogComment start="/\*" end="\*/" ***@***.*** ',
\              'syn match  verilogComment "//.*" ***@***.*** ',
\           ],
\    },

Hi~ I am trying to use your solution, but it doesn't work for me. The error info is in the following picture. Would you please help me figure it out?
image.png (view on web)
My configuration is:
image.png (view on web)
Thank your for your help in advance :)

netvolcano, 2023-12-23:
Have you installed the the verilog syntax plugin as below?
Plugin 'vhda/verilog_systemverilog.vim'

If it will not take effects after you installed the plugin, you need delete a line of "syn keyword verilogLabel       begin end fork join" from the verilog_systemverilog.vim file in the syntax directory.

netvolcano, 2023-12-23:
You need to modify the below statement in the verilog_systemverilog.vim file in the syntax directory (I think it is the root for your error messages):
#231 syn keyword verilogStatement  begin end 
syn keyword verilogStatement  begin end
[image]

real-bird-kun, 2023-12-23:
> You need to modify the below statement in the verilog_systemverilog.vim file in the syntax directory (I think it is the root for your error messages): # syn keyword verilogStatement begin end syn [...]

Cool, you're totally right. It works. Thanks a lot.
I am wondering how your vimrc looks like, because I think you must have a lot great config for writing veirlog/sv efficiently.
Anyway, thank you for your help in time :)