vhda/verilog_systemverilog.vim #207: System Verilog struct and class member auto completion

tianrui-wei, 2021-05-05:
Hi Vitor,

Many thanks for your excellent work on this plugin! I was trying it, and saw that sv structs and class still can't do autocompletion. I saw issue #150 is still open and got to wonder if there's some progress after that issues open. For reference, my TAGS file was generated with the command `ctags --extras=+q --fields=+i -n -R`

Hope you're having a great time with your family!

Thanks
Tianrui

vhda, 2021-08-28:
Hi Tianrui,

Sorry for the late reply, but I missed this request.
Unfortunately, free time has not been abundant, so I've not worked on that development.

Thanks for understanding,
Vitor

tianrui-wei, 2021-08-29:
Hi vitro,

It’s no problem at all. I really appreciate you writing this reply and your already fantastic vim plug-in. Best of luck with your other duties.

Best regards,
Tianrui

vhda, 2021-08-30:
Let me leave this open as a reminder.