vhda/verilog_systemverilog.vim #208: Coc-snippets doesn't trigger

arsoc, 2021-08-25:
Hi,

Thanks for a great job with this plugin. As far as syntax highlighting, folding and indentation I have no issues.

But it seems that verilog_systemverilog plugin doesn't work with coc-snippets. I did some very basic diagnostics for coc and it tells that snippets have been loaded correctly. It's just that they don't show up in completion menu, although other types of completion does.

Can you, please, investigate the issue or at least provide me with some directions of where to look and what to try? If any further information/testing is necessary, I'm ready to provide it as soon as I have free time.

Thanks in advance,
Artem

vhda, 2021-08-28:
You'll probably need to ask coc-snippets for support, because I don't know how it works.
Are you sure you are using `verilog_systemverilog` as the filetype trigger, instead of `verilog`?

arsoc, 2021-08-29:
Thanks for a quick reply.
Yeah, I've checked the filetype and it's set to `verilog_systemverilog`. 
Also, as I've already mentioned that snippets are detected and listed in `:CocList sources` command output but not triggered. It seems like something in `verilog_systemverilog` plugin conflicts with coc.nvim completion.

vhda, 2021-08-29:
On line 12 of file `ftplugin/verilog_systemverilog.vim` I'm setting `'omnifunc'`, which may interfere with COC. Could you please try commenting out that line and see if it improves anything?
If this is the source of the problem I can easily add an option to disable this feature.

arsoc, 2021-08-30:
Tried this one but with no luck. 
While doing some extra tweaking I noticed that snippets work when filetype is `systemverilog`, but not `verilog_systemverilog`. So, I extended snippets for `verilog_systemverilog` filetype with `systemverilog` filetype:
```
  "snippets.extends": {
    "verilog_systemverilog": ["systemverilog"]
  }
```
Everything works now with this quick workaround. Thanks for your time!

closed 2021-08-30 by arsoc