vhda/verilog_systemverilog.vim #209: Can I get this indent? if yes what settings do I need to use...

vineyg, 2022-02-14:
What I want is:

```
  assign abc = def
             & ghi
             & ( jkl
               | mno)
             & pqr;

```

what I am getting by default is (set shiftwidth=2):

```
  assign abc = def
  & ghi
  & ( jkl
    | mno)
               & pqr;

```

vhda, 2022-02-26:
Hi @vineyg,

While not exactly as in your example, by default the script should indent as following:
```systemverilog
 assign abc = def
              & ghi
              & ( jkl
                | mno)
              & pqr;
```
Please confirm that you have set `'autoindent'` Vim setting, which is required for automatic indentation to work.

Cheers,
Vitor

vineyg, 2022-02-28:
I did but still doesn't work the way you are describing
set autoindent
in .vimrc

vineyg, 2022-02-28:
I am running zsh shell not bash, does it have to do anything with that

vhda, 2022-03-01:
What does it show when you run `:set filetype`?

vineyg, 2022-03-01:
verilog_systemverilog

vineyg, 2022-03-01:
If I use operation end of the line it works fine:

`  assign abc = def &
               ghi &
               ( jkl |
                 mno) &
               pqr;
`

vhda, 2022-03-11:
Don't understand why you're getting different results than me.
Let's try the following:
```vim
:let b:verilog_verbose=1
:messages clear
```
Then select those 5 lines and press `=`.
Finally, copy&paste the contents of the window after you run:
```vim
:messages
```

vineyg, 2022-03-11:
When I wrote code it shows as:

`  assign abc = def
  & ghi
  & ( jkl
    | mno)
               & pqr;

`

When I select 5 lines and press =, it gets updated to 

`
  assign abc = def
               & ghi
               & ( jkl
                 | mno)
               & pqr;

`

Following are the messages:

Found possible end of instance on line 5 with level 2
GetContextIndent:4:     output  out
GetContextIndent:3:     input   in,
GetContextIndent:2:   (
GetContextIndent:1: module test
Inside a module.
GetContextIndent:7:   assign abc = def
Increasing indent for an open statement.
GetContextIndent:8:                & ghi
Increasing indent for an open statement.
GetContextIndent:7:   assign abc = def
GetContextIndent:9:                & ( jkl
Increasing indent for an open statement.
Inside a '()' block.
GetContextIndent:10:                  | mno)
Increasing indent for an open statement.
GetContextIndent:9:                & ( jkl
GetContextIndent:8:                & ghi
GetContextIndent:7:   assign abc = def
5 lines indented

vhda, 2022-03-21:
Branch [indent/indent_assign_on_symbol](https://github.com/vhda/verilog_systemverilog.vim/tree/indent/indent_assign_on_symbol) was created to add a new option to enable the behavior you expected. Please test it and report back.

vineyg, 2022-03-22:
this is what I get when I type same code:

```
  assign abc = def
  & ghi
  $ ( jkl
    | mno)
               & pqr;
```

and when I select text and press = it changes to 

```
  assign abc = def
               & ghi
  $ ( jkl
    | mno)
               & pqr;

```

vhda, 2022-03-23:
What does this show?
```vim
:set indentexpr?
```
Also, have you tried disabling all addons except this one?

vineyg, 2022-03-24:
What addons are you referring to?

vhda, 2022-03-24:
I am unable to replicate your behavior, so I'm suspecting it's either a difference in vim's settings or a plugin that's modifying the indent behavior.
For the first possibility I am trying to confirm that your `indentexpr` setting is correct. For the second the only way is to remove all plugins, except this one.

vineyg, 2022-03-29:
`:set indentexpr?`

shows

indentexpr=GetVerilogSystemVerilogIndent()

and I don't think I am using any other plugin...

vhda, 2022-09-11:
Hi @vineyg,
Completely forgot about this issue, sorry.
Do you have any updates?

vineyg, 2022-09-11:
Still same issue as mentioned above

vhda, 2022-09-12:
On the third line you have a dollar (`$`) instead of an ampersand (`&`).

vineyg, 2022-09-12:
That might be a typo but irrespective the behavior remains same. When you
use this extension, what output you get?

vhda, 2022-09-12:
I get this by default:
```systemverilog
assign abc = def
             & ghi
             & ( jkl
               | mno)
             & pqr;
```
And with `g:verilog_indent_assign_on_symbol=1`:
```systemverilog
assign abc = def
           & ghi
           & ( jkl
             | mno)
           & pqr;
```

vhda, 2022-09-17:
Have you confirmed these results?

pat-lall, 2022-11-19:
@vhda I tested and confirmed your results.  I see the same behavior as you.

vhda, 2022-11-20:
Thanks @pat-lall

closed 2022-11-20 by vhda