vhda/verilog_systemverilog.vim #215: neovim 0.72， 1st open .v file,  variable  g：verilog_syntax  can not find error as below

microway199x, 2022-09-11:
when use neovim 0.72， 1st open .v file， the variable, g:verilog_syntax can not find error as below, any one can fix it??

[image]

vhda, 2022-09-11:
Hi @microway199x,
I'm pretty sure something is broken in how you are setting up plugins in your Neovim configuration.
I've just installed Neovim and setup its init.vim to load my .vimrc and did not find any issues.

microway199x, 2022-09-12:
> 

thanks vhda reply：
I think I know why fails now；

my case scenario: in windows 10, add quick open menu @ mouse right-click menu (regedit command: path/nvim-qt.exe %1), then open  .v file with right-click;

for Nvim load file sequence different from GVIM, so Gvim work pretty, but Nvim work with error alarm;

global variable g:verilog_syntax defined @plugin/verilog_systemverilog.vim file ,but the execute time later than @syntax/verilog_systemverilog.vim and @indent/verlog/verilog_systemverilog.vim, 

so I cut the variable define to ftplugin/verilog_systemverilog.vim， and it works now；

[image]

[image]

thanks again for your great share；

vhda, 2022-09-12:
Loading the plugin folders first it's the normal vim behavior, to which I'm pretty sure neovim is compatible. How are you loading the plugins?

microway199x, 2022-09-12:
For all plugins I used， I just put it @. appdata/nvim  folder.
 For neovim just copy all files under vimfiles/ direct.  
if sequence loaded is right， the error is nonlogical.

Personally，put some code to ftplugin and make it work right first. maybe for other user， it work correct.

thanks

vhda, 2022-09-12:
I was wrong: https://neovim.io/doc/user/starting.html#initialization
Apparently Neovim uses a different initialization order, where ftplugin and syntax are loaded before plugins.
I'll need some time to see what would be the best approach here.

vhda, 2022-09-12:
I'm unable to reproduce the errors you are seeing.
This is my script load order, with only this single plugin in nvim local config:
```
:scriptnames                                                                                                                                          
  1: /usr/share/nvim/runtime/ftplugin.vim
  2: /usr/share/nvim/runtime/indent.vim
  3: /usr/share/nvim/runtime/filetype.lua
  4: /usr/share/nvim/runtime/filetype.vim
  5: ~/.config/nvim/ftdetect/verilog_systemverilog.vim
  6: /usr/share/nvim/runtime/syntax/syntax.vim
  7: /usr/share/nvim/runtime/syntax/synload.vim
  8: ~/.config/nvim/plugin/verilog_systemverilog.vim
  9: /usr/share/nvim/runtime/plugin/gzip.vim
 10: /usr/share/nvim/runtime/plugin/health.vim
 11: /usr/share/nvim/runtime/plugin/man.vim
 12: /usr/share/nvim/runtime/plugin/matchit.vim
 13: /usr/share/nvim/runtime/pack/dist/opt/matchit/plugin/matchit.vim
 14: /usr/share/nvim/runtime/plugin/matchparen.vim
 15: /usr/share/nvim/runtime/plugin/netrwPlugin.vim
 16: /usr/share/nvim/runtime/plugin/rplugin.vim
 17: /usr/share/nvim/runtime/plugin/shada.vim
 18: /usr/share/nvim/runtime/plugin/spellfile.vim
 19: /usr/share/nvim/runtime/plugin/tarPlugin.vim
 20: /usr/share/nvim/runtime/plugin/tohtml.vim
 21: /usr/share/nvim/runtime/plugin/tutor.vim
 22: /usr/share/nvim/runtime/plugin/zipPlugin.vim
 23: /usr/share/nvim/runtime/scripts.vim
 24: ~/.config/nvim/ftplugin/verilog_systemverilog.vim
 25: ~/.config/nvim/indent/verilog_systemverilog.vim
 26: ~/.config/nvim/syntax/verilog_systemverilog.vim
 27: ~/.config/nvim/autoload/verilog_systemverilog.vim
```
The ftdetect and plugin folders are the first to be loaded, and only at the end (after the buffer is loaded, I assume) will the remaining folders be loaded.
Could you try opening the Verilog file with nvim -u NONE?

closed 2022-09-17 by vhda