vhda/verilog_systemverilog.vim #227: Verilog_syntax error

vinayakkp, 2023-02-22:
Hi,

When I am trying to load a Verilog header file and I see this error:

Error detected while processing $HOME/.vimrc[1434]..$HOME/tools/share/vim/vim90/syntax/syntax.vim[43]..BufRead Autocommands for "*.vh"..FileType Autocommands for "*"..function <SNR>9_LoadIndent[14]..script $HOME/.vim/bundle/verilog_systemverilog.vim/indent/verilog_systemverilog.vim:
line   45:
E121: Undefined variable: g:verilog_syntax
line   46:
E121: Undefined variable: g:verilog_syntax
line   47:
E121: Undefined variable: g:verilog_syntax
Error detected while processing $HOME/.vimrc[1434]../$HOME/tools/share/vim/vim90/syntax/syntax.vim[43]..BufRead Autocommands for "*.vh"..FileType Autocommands for "*"..Syntax Autocommands for "*"..function <SNR>4_SynSet[25]..script $HOME/.vim/bundle/verilog_systemverilog.vim/syntax/verilog_systemverilog.vim:
line  246:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>20_SyntaxCreate
line  341:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>20_SyntaxCreate

I am using the latest code from master as of Feb 22nd. Am I missing something?

Thx,

vhda, 2023-02-23:
Looking at those paths I can see your are copying the files of this plugin over vim's standard installation, which is incorrect.
I suggest you use a plugin manager like vim-plug.

vinayakkp, 2023-02-24:
Hi, 

I am using Vundle plugin manager and have following code:

call vundle#begin()
Plugin 'gmarik/Vundle.vim'
Plugin 'vhda/verilog_systemverilog.vim'
call vundle#end()

I am using latest VIM 9.0

vinayakkp, 2023-02-24:
Thanks for response. I did bunch of cleanup of old plugins and it now seems to work. If I can reproduce the old issue again, will update this ticket.

closed 2023-03-05 by vhda