vhda/verilog_systemverilog.vim #228: Undefined variable: g:verilog_syntax

bpatel2525, 2023-03-04:
Error detected while processing /home/bhargavrajp/.vimrc[3]../home/utils/vim-9.0.0630/share/vim/vim90/syntax/syntax.vim[43]..BufRead Autocommands for "*.sv"..FileType Autocommands for "*"..Syntax Autocommands for "*"..function <SNR>4_SynSet[25]..script /home/bhargavrajp/.vim/bundle/verilog_systemverilog.vim/syntax/verilog_systemverilog.vim: 
line  246: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate 
line  341: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate 
line  246: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate 
line  341: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate 
line  246: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate 
line  341: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate

vhda, 2023-03-05:
Had another report of similar issue a few days back in #227, even though nothing significant has changed in this regard for years. Original requester of that issue reported that everything started working after cleaning up their plugins. Would you mind trying that on your side?
What plugin manager are you using?

sqlmap3, 2024-05-11:
8.2 and 7.4 is ok。but vim9.0 is not work.

sqlmap3, 2024-05-11:
Error detected while processing /home/xxx/.vimrc[4]../ic/tools/vim/9.0/share/vim/vim90/syntax/syntax.vim[43]..BufRead Autocommands for "*.sv"..FileType Autocommands for "*"..Syntax Autocommands for "*"..function <SNR>4_SynSet[25]..script /home/xx/.vim/bundle/verilog_systemverilog/syntax/verilog_systemverilog.vim

sqlmap3, 2024-05-11:
line  246:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>14_SyntaxCreate
line  341:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>14_SyntaxCreate

bpatel2525, 2025-01-30:
Sorry was away for few months and than forgot! 

I can confirm sqlmap3 observation that 8.2 and 7.4 is ok but 9.0 or 9.1 don't work. Thanks sqlmap3.

Moved back to 8.2 and things are fine.

keeeeymann, 2025-06-18:
I encountered this lately with nvim, and my guess is that it relates to the LOAD ORDER between the vim-bundled `systemverilog` ftplugin and our plugin. If `systemverilog` ftplugin is not loaded yet, the issue happens.

 to reproduce the issue:

1. edit `ftdetect/verilog_systemverilog.vim`:
    replace `setfiletype` command with `set filetype=` of the autocmd

2. open SV file via command line:
    `$ nvim some.sv`

3. nvim now complains `g:verilog_syntax` missing on startup.

4. if we re-start a bare nvim and open the file afterwards:
    `$ nvim`
    `:e some.sv<CR>`

5. then nvim is happy to open the file.

Without the ftdetect edit, nvim always uses its bundled `systemverilog` ftplugin. (And that is my original intention of the edit.)