vhda/verilog_systemverilog.vim #229: Intendation is slow when inserting a new line into the thousands signal interface

netvolcano, 2023-03-06:
I found an issue with the plugin of 3.0 version. When I have more 2 thousands of interface signals, there will be a long time waiting to finish GetContextIndent function (I used a profile log file to confirm this). I tried the current master version of the plugin also, but it can not slove the problem. 

 I found this issue similar to the issue [ Intendation at end of medium-large files is slow #190 ].

I suggest there will be a max limit of the scaned line number in the GetContextIndent function.

netvolcano, 2023-03-06:
[image]

vhda, 2023-05-10:
The current implementation of the indentation script focuses on correctness of indentation, which would become broken with such limit.
This limit can be implemented, but first I'd like to better understand your scenario to make sure there are no alternatives.
What are you doing when this slowness occurs? Are you indenting the complete file, or simply starting a new line?
In case of the latter, would it be possible to show the immediate context before this new line? I understand you will not be able to share the 2k lines file, but I'd like to understand if there's something we could use to obtain an indentation value faster.
Thanks!

netvolcano, 2023-05-11:
When I opened a verilog file with more than 2 thousands of interface signals, I wanted to insert a new line to add a new signal, Vim would waiting long time to response (it seemed to scan too many lines to determine the correct context). I tried to add a limit to constrain the max scan lines to speed this. I think you can add a global variable to implement this (i.e. It is better to record the current context into a variable. When reached the max scaned line number but cannot determine the context, you can use the recorded context instead).