vhda/verilog_systemverilog.vim #230: This plugin breaks verilog_emacsauto.vim

ben-k, 2023-04-26:
When I install this plugin, [verilog_emacsauto.vim](https://www.vim.org/scripts/script.php?script_id=1875) stops working for me.  Is there a recommended way to use this plugin and still allow Emacs verilog mode functionality within vim?

vhda, 2023-05-10:
I'm sorry, but I can't see anything in that plugin that would become broken my this plugin.
Is it possible you are installing the plugins incorrectly?

ben-k, 2023-05-10:
I'm sure it's possible :)

If you're unable to reproduce on your end, feel free to close this... I don't have bandwidth to diagnose further.

vhda, 2023-05-10:
I'm not an Emacs user, don't even have it installed, so I didn't do much debug to be honest.
But while writing this I decided to at least install it and modify it to echo stuff instead of calling Emacs.
Then I noticed it is being installed as an ftplugin, which makes it special!
In the name verilog_emacsauto.vim the first word represents the filetype, but in this plugin the filetype is verilog_systemverilog and so emacsauto will not be loaded.
Please move verilog_emacsauto.vim into folder .vim/ftplugin/verilog_systemverilog/ and try again.

ben-k, 2023-05-12:
Works perfectly, thanks for looking into it!  (I also never use emacs, except for this one purpose 🙂)

closed 2023-05-12 by ben-k