vhda/verilog_systemverilog.vim #231: [BUG] macros breaking indentation

miguel9554, 2023-06-30:
When using macros inside other syntax elements with their own indenting (module instantiation, fucntions, contraints etc) the indentation is broken, see following example

```systemverilog
class my_class;

    int my_var;

    constraint reg_addr_c {
        `ifndef MY_MACRO
            my_var == 2;
        `else
            my_var == 3;
        `endif
    }

        function new();
        endfunction

endclass
````

The `function` declaration should be at the level of constraint again, but it is at the level of the macro block.

vhda, 2023-06-30:
I sometimes feel that certain SystemVerilog features were developed by JavaScript engineers...
I'm pretty sure the problem is related with that semicolon inside the brackets and the expression not ending in a semicolon, but will need sometime to investigate (sorry, real life TM).

ThunderMikey, 2023-06-30:
The same if I do

```systemverilog
`uvm_analysis_imp_decl(_w)

  class xxx;
  endclass
```

vhda, 2023-06-30:
UVM is also done by JavaScript engineers 😝

miguel9554, 2023-06-30:
Hey, I don't think it has to do with the semicolon inside constraints block, that seems to work OK. Some examples

Removing macro, using constraint block per correct SV syntax works OK

```systemverilog
class my_class;

    int my_var;

    constraint reg_addr_c {
        my_var == 2;
    }

    function new();
    endfunction

endclass
```

You can also try not including the semicolon, or including it after the brace (even though it doesn't compile, just to test the vim plugin) and indentation works OK.

The problem happens when you have the macro inside the block

miguel9554, 2023-06-30:
> The same if I do
> 
> ```systemverilog
> `uvm_analysis_imp_decl(_w)
> 
>   class xxx;
>   endclass
> ```

I've copied that into an empty file and could not reproduce, it worked OK

ThunderMikey, 2023-07-03:
Oops, sorry. My bad, it turns about there is a `ifndef` which caused the indentation. I guess this is feature, not a bug (TM).

```systemverilog
`ifndef XXX
  `define XXX
  `uvm_analysis_imp_decl(_w)
```

vhda, 2023-07-04:
It's always a good day when what we thought was a bug in fact isn't!
Shall I close this issue then?

miguel9554, 2023-07-04:
No, don't close, what I reported initially **is** a bug. What isn't, is what @ThunderMikey reported.

puliget, 2023-08-23:
Using macro inside constraints, is working fine. You code is working. Check this link and run the code https://edaplayground.com/x/MHuL

Read this document:
[image]

miguel9554, 2023-08-24:
@puliget Of course macros work inside constraints, who said they didn't? I'm reporting that this particular VIM indentation file does not handle well indentation for those cases.

The document you linked is part of the verilog 2001 standard, which has no constraints in it... I don't get your point

puliget, 2023-08-25:
class my_class;

  rand  int my_var;

    constraint reg_addr_c {
        `ifndef MY_MACRO
            my_var == 2;
        `else
            my_var == 3;
        `endif
    }

        function new();
        endfunction

endclass

module tb;
  my_class cls;
  
  initial begin
    cls=new();
    cls.randomize();
    $display("my_var=%d",cls.my_var);
  end
endmodule

puliget, 2023-08-25:
It is working, check once link :https://edaplayground.com/x/MHuL
and run code in eda playground

miguel9554, 2023-08-26:
@puliget again, I agree on macros inside constraints being valid systemverilog constructs, I never said contrary. In fact, I opened this issue because I was making use of the feature.

The bug is reported for a VIM syntax file, you are not getting the point. How can the problem be reproduced in eda playground if it is specific to the VIM editor?

puliget, 2023-08-26:
Eda playground for code  editing and runing.many tools are available in eda playground for run codes . I use the mentor questa to run my code in eda playground and i shared that link to you.   Vim editor is used for coding editing.on which tool,you are running the code ..

puliget, 2023-08-26:
Never learn , based on tools. Learn based on the syntax only.

miguel9554, 2023-08-27:
@puliget  you really are not understanding the issue here

vhda, 2024-01-23:
@miguel9554 could I kindly ask you to take a test run of https://github.com/vhda/verilog_systemverilog.vim/tree/indent/skip_bracket_blocks ?

vhda, 2024-10-13:
@miguel9554 Any updates?