vhda/verilog_systemverilog.vim #234: Add commentstring for vim commentary

mpkopec, 2024-01-18:
Hi!
I (and probably not only I) use vim-commentary by T. Pope to comment stuff out in the code, this plugin relies on the `commentstring` setting, which is local to buffer, and in my opinion should be defined withing the Verilog syntax definition, but I am a complete noob in VimScript and plugin writing. Could you guide me on where would be the best place to put a definition as in [my vimrc](https://github.com/mpkopec/nvim-config/blob/529efdaba5aa49f0d3fb37d29f590b4cbd892cb8/plugconf/verilog_systemverilog.vim#L7) in this plugin?  I would like to submit a PR, but have no idea, where to put it not to mess anything up in the grand scheme of things.

Thanks a lot!

vhda, 2024-01-22:
Hi @mpkopec,
Unfortunately that option only supports a single format of comments.
Given that its default is `"/*%s*/"` and that this is a valid comment format in Verilog/SystemVerilog, then it makes little sense for me to change this centrally (even though `//` is also my preferred comment format).
There is also this issue in vim-commentary that appears to point this as being a limitation in that plugin https://github.com/tpope/vim-commentary/issues/44.
All together, at this point I'd say that this is really a matter of personal preference and that you are using the correct approach to configure it according to yours.
Would you agree?
Kind regards,
Vitor

mpkopec, 2024-01-23:
Thanks a lot for the reply. Indeed, it seems, that putting it inside the plugin and not leaving it up to the user is not optimal. Thanks a lot!

closed 2024-01-23 by mpkopec

vhda, 2024-01-23:
Thanks for understanding and please don't hesitate in proposing more changes.