vhda/verilog_systemverilog.vim #235: Folding works on some files

gyh1997127, 2024-01-22:
Hi, I'm noticing that the syntax folding on works on some of my files. Can you please advise how to go about debugging this? Example: 
[image]
[image]

Apologies for not being able to include unblurred examples, but you can see the folding works on one file but not the other.

vhda, 2024-01-22:
Hi @gyh1997127,
Could I kindly ask you to check if folding works after removing the `` `uvm_object_utils()`` line?
It is possible that the missing `;` is breaking a basic end-of-expression marking that makes it really hard sometimes to cover in syntax expressions.

gyh1997127, 2024-01-22:
I didn't see any changes after removing `uvm_object_utils()`. I don't think it's the culprit - this statement is also present in the first screenshot on line 15

vhda, 2024-01-23:
Could I kindly ask you to try to replicate the problem using a smaller test case? It's really difficult to understand what is happening without the complete context.