vhda/verilog_systemverilog.vim #237: Undefined variable: g:verilog_syntax

MinhTran0911, 2024-03-19:
I just installed this plugin using Vim Plug by:
1. Download the entire repo as zip -> Extract -> Copy contents into ~/.vim
2. Add the following code to ~/.vimrc:

```vim
call plug#begin()
Plug 'vhda/verilog_systemverilog.vim'
call plug#end()
```
3. Run

```shell
%vim +PlugInstall +qall
```

4. Then I open a SystemVerilog file (*.sv extension) and get the following error

```
Error detected while processing /usr/share/vim/vimrc[25]../usr/share/vim/vim82/syntax/syntax.vim[42]..BufRead Autocommands for "*.sv"..FileType Autocommands for "*"..Syntax Autocommands for "*"..function <SNR>4_SynSet[25]..script /home/minh/.vim/syntax/verilog_systemverilog.vim:
line  246:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>10_SyntaxCreate
line  341:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>10_SyntaxCreate
line  246:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>10_SyntaxCreate
line  341:
E121: Undefined variable: g:verilog_syntax
E116: Invalid arguments for function <SNR>10_SyntaxCreate
Press ENTER or type command to continue
```

5. I saw a similar issues on this repo and try all recommended debugging steps including comment out the entirety of my .vimrc but still not working.
Can you help to check this issue?

vhda, 2024-03-22:
Step 1 should not be needed if you are using VimPlug to install this plugin.
Please move your ~/.vim to another location and re-run steps 2 to 4 and report back if it helps.
If it still doesn't work, please share the output of `:echo runtimepath`.
(Note: you should omit your $HOME location to avoid sharing your username online. This is also valid in your original post.)

MinhTran0911, 2024-03-24:
So it turns out that for Vim 8.2, I can only install this using VimPlug. Manual installation does not work.
On Vim 7.2 manual installation doesn't seem to be a problem though.
Thank you for your help.

closed 2024-03-24 by MinhTran0911