vhda/verilog_systemverilog.vim #239: indentation with virtual function/task and fork/join

albasili, 2024-05-03:
I'm having serious indentation issues with 'virtual function/task' as well as with 'fork/join' blocks, the result is quite weird:

This is the function I have:
```
    /* fsm_transition
    When fsm state makes transition.
     */
    virtual function void fsm_transition (
        int                id,
        fsm_state_e from,
        fsm_state_e to
    );
```

when highlighted and hit "=" for indenting correctly than I have the following:
```
    /* fsm_transition
    When fsm state makes transition
        */
       virtual function void fsm_transition (
           int                id,
           fsm_state_e from,
           fsm_state_e to
       );
```

I have also the 'Yggdroot/indentLine' plugin to show the indent line and everything seems to be messed up.
My settings for systemverilog filetypes are the following:
```
" Verilog and SystemVerilog tab settings
au BufNewFile,BufRead *.sv,*.v,*.svh,*.vh,*.vams setlocal ts=4 sts=4 sw=4 tw=120
" Highlighting for .inc files
au BufNewFile,BufRead *.sv,*.v,*.svh,*.vh,*.vams,*.sv.inc set ft=systemverilog
```

Any pointer to where I should be looking?

vhda, 2024-05-03:
Please note that this plugin's filetype is `verilog_systemverilog`. Are you sure you are using it?

albasili, 2024-05-03:
holy crap! Sorry about that, I did not realize the filetype was not correct.
Now it works as a charm!

closed 2024-05-03 by albasili