vhda/verilog_systemverilog.vim #241: tagbar need add other labels

sqlmap3, 2024-05-11:
According to  web. []([https://docs.ctags.io/en/latest/man/ctags-lang-verilog.7.html],tagbar) need add other labels.
```
ctags --list-kinds-full=SystemVerilog
#LETTER NAME       ENABLED REFONLY NROLES MASTER DESCRIPTION
A       assert     yes     no      0      NONE   assertions (assert, assume, cover, restrict)
C       class      yes     no      0      NONE   classes
E       enum       yes     no      0      NONE   enumerators
H       checker    yes     no      0      NONE   checkers
I       interface  yes     no      0      NONE   interfaces
K       package    yes     no      0      NONE   packages
L       clocking   yes     no      0      NONE   clocking
M       modport    yes     no      0      NONE   modports
N       nettype    yes     no      0      NONE   nettype declarations
O       constraint yes     no      0      NONE   constraints
P       program    yes     no      0      NONE   programs
Q       prototype  no      no      0      NONE   prototypes (extern, pure)
R       property   yes     no      0      NONE   properties
S       struct     yes     no      0      NONE   structs and unions
T       typedef    yes     no      0      NONE   type declarations
V       covergroup yes     no      0      NONE   covergroups
b       block      yes     no      0      NONE   blocks (begin, fork)
c       constant   yes     no      0      NONE   constants (parameter, specparam, enum values)
d       define     yes     no      0      NONE   text macros
e       event      yes     no      0      NONE   events
f       function   yes     no      0      NONE   functions
i       instance   yes     no      0      NONE   instances of module or interface
l       ifclass    yes     no      0      NONE   interface class
m       module     yes     no      0      NONE   modules
n       net        yes     no      0      NONE   net data types
p       port       yes     no      0      NONE   ports
q       sequence   yes     no      0      NONE   sequences
r       register   yes     no      0      NONE   variable data types
t       task       yes     no      0      NONE   tasks
w       member     yes     no      0      NONE   struct and union members
```

sqlmap3, 2024-05-11:
[web](https://docs.ctags.io/en/latest/man/ctags-lang-verilog.7.html)

vhda, 2024-07-20:
It's been a while since I last updated Universal Ctags.
Would you be so kind and edit this in plugin/verilog_systemverilog.vim to match your version of Universal Ctags and create a PR for this?

```vim
    let g:tagbar_type_verilog_systemverilog = {
        \ 'ctagstype'   : 'SystemVerilog',
        \ 'kinds'       : [
            \ 'b:blocks:1:1',
            \ 'c:constants:1:0',
            \ 'e:events:1:0',
            \ 'f:functions:1:1',
            \ 'm:modules:0:1',
            \ 'n:nets:1:0',
            \ 'p:ports:1:0',
            \ 'r:registers:1:0',
            \ 't:tasks:1:1',
            \ 'A:assertions:1:1',
            \ 'C:classes:0:1',
            \ 'V:covergroups:0:1',
            \ 'I:interfaces:0:1',
            \ 'M:modport:0:1',
            \ 'K:packages:0:1',
            \ 'P:programs:0:1',
            \ 'R:properties:0:1',
            \ 'T:typedefs:0:1'
        \ ],
        \ 'sro'         : '.',
        \ 'kind2scope'  : {
            \ 'm' : 'module',
            \ 'b' : 'block',
            \ 't' : 'task',
            \ 'f' : 'function',
            \ 'C' : 'class',
            \ 'V' : 'covergroup',
            \ 'I' : 'interface',
            \ 'K' : 'package',
            \ 'P' : 'program',
            \ 'R' : 'property'
        \ },
    \ }
```
Thanks!

sqlmap3, 2024-07-24:
I have been modified tagbar for some days .