vhda/verilog_systemverilog.vim #243: vim become very slow when there is a long define

sin-x, 2024-08-03:
Hi, 
When the cursor moves within the long define block, Vim becomes very slow. But when the cursor moves outside the range of the define, it becomes normal.  

I have some files which have long defines. When move the cursor with these defines, it become very slow. So could you please help me find the problem?

The define like this but longer than this.
```
`define CEATE_SEQUSER(name, seqtyepe, seqrname) \
  if *** \
   *** \
   
```

Aprilskysky, 2024-08-09:
i have the sam issue, when i use long define, neovim will become very slow.

vhda, 2024-08-09:
I've experienced this when using syntax folding, hence my recommendation to use FastFold plugin at the end of README.md in the front page. Have you tried this?

Aprilskysky, 2024-08-10:
Thanks for your reply. I am in normal mode, long define will be very slow. But your recommendation to use FastFold plugin at the end of README.md is in insert mode. The FastFold plugin also overcomes this issue in normal mode?

sin-x, 2024-08-10:
Even though I set the foldmethod to marker, the problem still persists. When i use hjkl to move cursor, vim becomes very slow. But when I use Ctrl-f to scroll through pages, vim does't become slow.

sin-x, 2024-08-10:
sorry for the accidental operation.

Aprilskysky, 2024-08-27:
"let g:verilog_quick_syntax = 1" will let long define don't become very slow. But automatic indentation will be desiabled.

antoine-axelera-ai, 2025-02-19:
@Aprilskysky `let g:verilog_quick_syntax = 1` didn't fix the slowness for me.