vhda/verilog_systemverilog.vim #245: indent.vim cannot recognize randcase/endcase correctly

bobo-acount, 2024-10-04:
hi sir,
when using randcase/endcase pair, this script could not do indent correctly (`endcase` will not keep indent as `randcase`) as follows.
from the indent.vim, i thought it maybe caused that `v_block_start` does not add `randcase` statement.
i have tried locally to add `randcase` support, but my fix affects regular `case()/endcase` indent.
too complicated for me since `randcase` has no `()` but `case` requires `()`

```verilog
     randcase
endcase
````

thanks a lot for your script

vhda, 2024-10-13:
Please confirm if latest code fixed the issue you were seeing and reopen if that's not the case.
Thanks!

closed 2024-10-13 by vhda

bobo-acount, 2024-10-25:
sorry for late reply
latest code partially fix the issue but still has tiny issue:
after  hit enter key after ```endcase```, the indent seems to be reset, and cursor comes back to beginning of the line, not aligns with ```endcase```, which is the expected location

thanks!

bobo-acount, 2024-10-25:
by the way, i checked test/indent.sv which contains ```randcase/endcase``` test. and ```randcase``` starts at the begging with the line by coincidence. if you wrap ```randcase/endcase``` into another ```begin/end``` block, and type some statement following ```endcase```, the issue could be replicated

PR #246 (2024-10-13): Indent: Improve support for randcase
* Include 'randcase' keyword in b:match_words
* Drop s:vlog_block_start/end from indent/verilog_systemverilog.vim since they are no longer in use
* Add simple indent test for randcase

Fixes #245.
commit 62fe7bdf (2024-10-13): Indent: Improve support for randcase

* Include 'randcase' keyword in b:match_words
* Drop s:vlog_block_start/end from indent/verilog_systemverilog.vim
  since they are no longer in use
* Add simple indent test for randcase

Fixes #245.